NachtRaveVL/PCA9685-Arduino #7: unterminated #ifndef - PCA9685.cpp

vaiko, 2018-02-10:
I’m C++ noob and taking my first steps with the Arduino (IDE 1.8.5).
I’m running into an error compiling the “SimpleExample” provided with the library version 1.2.10:

```
:\Users\Mein Name\Documents\Arduino\libraries\PCA9685-Arduino-master\PCA9685.cpp:600:0: error: unterminated #else
 #ifdef PCA9685_ENABLE_SOFTWARE_I2C
 ^

C:\Users\Mein Name\Documents\Arduino\libraries\PCA9685-Arduino-master\PCA9685.cpp:588:0: error: unterminated #else
 #ifdef PCA9685_ENABLE_SOFTWARE_I2C
 ^

C:\Users\Mein Name\Documents\Arduino\libraries\PCA9685-Arduino-master\PCA9685.cpp:264:0: error: unterminated #else
 #ifdef PCA9685_ENABLE_SOFTWARE_I2C
 ^

C:\Users\Mein Name\Documents\Arduino\libraries\PCA9685-Arduino-master\PCA9685.cpp:222:0: error: unterminated #ifndef
 #ifndef PCA9685_EXCLUDE_EXT_FUNC
 ^

exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Uno.
```

Do you have an idea what I could be doing wrong? Counting the closing #endif (45) for the opening # ifdef (40) and ifndef (9) in PCA9685.cpp indicates a mismatch but I’ve just started learning so I don’t know if my counting is valid.

vaiko, 2018-02-10:
Going back to commit ac4a15d34499375e8993ea5b907f2fd5ca1f1b8b fixed the issue for me but it also removes the latest bugfixes.

Nzbuu, 2018-02-12:
I get this problem too

NachtRaveVL, 2018-02-12:
Thanks for letting me know. #else got sneaked in there when it should had been #endif.

Latest commit should fix this.

closed 2018-02-12 by NachtRaveVL

Nzbuu, 2018-02-13:
Working now. Thanks 👍