NachtRaveVL/PCA9685-Arduino #13: Examples won't compile on ESP32 ?

CRprog, 2020-03-06:
I'am using an ESP32 and want to use the setChannelsPWM to set 6 Servos at one I2C-write.
But the BatchingExample producing compile-errors:
_no matching function for call to 'PCA9685::setChannelsPWM(int, int, word [12])'_
and also other examples fail:
_S:\arduino-1.8.10_ESP32\portable\sketchbook\libraries\PCA9685-Arduino\PCA9685.cpp:208:45: error: 'BUFFER_LENGTH' was not declared in this scope
         int maxChannels = min(numChannels, (BUFFER_LENGTH - 1) / 4);_
Any Idea ?

NachtRaveVL, 2020-03-07:
I left this as a comment around its usage:

// In avr/libraries/Wire.h and avr/libraries/utility/twi.h, BUFFER_LENGTH controls
// how many channels can be written at once. 

Since you're not using the AVR libraries (you're using ESP32) you'll probably have to self-define this.

Still may be worth coding in a default to use in any case.

CRprog, 2020-03-07:
Thank you for the hint, For me, I've added the marked (#) lines here in PCA9685.h
` .....`
`#define PCA9685_MIN_CHANNEL         0`
`#define PCA9685_MAX_CHANNEL         15`
`#define PCA9685_CHANNEL_COUNT       16`
`#ifndef BUFFER_LENGTH`     (#)
` #define BUFFER_LENGTH 32`    (#)
`#endif`    (#)
` .....`
Now it compiles. But maybe there's a better solution at another place ....
Btw.: If you maybe update sometime, think about adding a keywords.txt (?)

closed 2020-03-07 by CRprog

NachtRaveVL, 2020-03-13:
Fixed on version 1.2.11. Pushed to master. No worries =). It was a good catch for an odd side case. Let me know if the new version helps or not. =)

CRprog, 2020-03-15:
Runs fine now 'out of the box' also in the ESP32 environment. Good job and have a nice day.