NachtRaveVL/PCA9685-Arduino #17: [Minor improvement] Adding keywords.txt to library

CRprog, 2020-10-17:
Hi!
Is there a reason not to include a keywords.txt file in the library?
I've created my own to avoid typos (and because it looks nicer ;-) :

PCA9685		KEYWORD1

init	KEYWORD2
initAsProxyAddresser	KEYWORD2
enableAllCallAddress	KEYWORD2
resetDevices	KEYWORD2
setPWMFrequency	KEYWORD2
setChannelPWM	KEYWORD2
getChannelPWM	KEYWORD2
setChannelsPWM	KEYWORD2
printModuleInfo	KEYWORD2
setChannelOff	KEYWORD2
setPWMFreqServo	KEYWORD2
getChannelPWM	KEYWORD2
pwmForAngle	KEYWORD2

But maybe "Methods and Functions" (KEYWORD2) or "Constants" (LITERAL1) are incomplete.

Regards.

NachtRaveVL, 2020-10-20:
That's an excellent suggestion, I'll see about adding such.

NachtRaveVL, 2020-10-24:
Will be in next version.

closed 2020-10-24 by NachtRaveVL