NachtRaveVL/PCA9685-Arduino #19: Start with predefined servo position - for example stored in eeprom

gryzli133, 2021-05-18:
first of all thank you for your library, it helped me a lot in my smart home projects ;-) Now I'm doing servo controlled ventilation flaps for my house ventilation system. It would be nice to store last value, and then start with last stored value, to avoid unnecessary movement on system restart (f.e. nights in bedroom you can really hear the movements). Is this possible with your library to avoid any movements before first set of setChannelPWM()? Eventually initialize with predefined position?

gryzli133, 2021-05-18:
Sorry, I think it was just fault in my sketch...

closed 2021-05-18 by gryzli133

NachtRaveVL, 2021-05-18:
Oh no worries whatsoever! Thanks for participating in the discussion. =)

I think that it's probably not something that is supported, however, just in that during a power reset the registers on the PCA9685 are cleared. You would at least have to write to the registers, but whatever the PCA9685 does under the hood would be the shortcoming there.

Glad you figured it out tho! ^_^